PantsForBirds/adsbee #42: Needs reset to reconnect to wifi

makrsmark, 2024-12-05:
My power went out and the adsbee came up before my wifi. As a result, it stopped feeding until i noticed. Pressing the reset button fixed everything (yay!). It would be nice to continually try connecting until successful.

CoolNamesAllTaken, 2024-12-05:
Thanks for flagging this! I currently have it set up to try a WiFi connection, then give up after a while, since I assumed that people having an ADSBee configured as a WiFi station and then taking it away from their home network would be a more common occurrence than what you described. After thinking about it some more, I think that the current implementation will be a problem, as most WiFI based feeders will likely encounter a power outage and will reboot faster than the WiFI network comes up.

I agree that the network behavior should be modified ot retry joining the WiFi network continuously until a connection is made.

makrsmark, 2024-12-06:
Cool. If you're concerned about it wasting cycles as a remote device, can the timeout at least be longer? Id have to guess 15-30min would be enough

CoolNamesAllTaken, 2024-12-06:
That sounds completely reasonable to me!

CoolNamesAllTaken, 2024-12-14:
This should be fixed by https://github.com/CoolNamesAllTaken/adsbee/pull/49

Still needs some dedicated testing, but network reconnection at least works on Ethernet (easier to plug / unplug than my home WiFi network which uses a number of different APs). The same logic should be working on WiFi but I haven't yet gotten around to testing with a dedicated access point.

I will close this issue for now, if you notice the behavior again (or have a chance to test with a dedicated access point), please reopen it!

closed 2024-12-14 by CoolNamesAllTaken