PantsForBirds/adsbee #43: FR: Ability to set hostname

kquinsland, 2024-12-05:
Right now, the hostname is deterministic (yay!) but not customizable.

It would be nice to have a way to customize the hostname for regular networking / and mDNS purposes.

I don't have my adsbee in front of me right now, so going from memory, the hostname for a device with MAC `00:02:DE:AD:BE:EF` ends up being something like:

```
adsbee-BEEF
```

Ideally there would be a way to set this through the web UI like so:

```html
<!DOCTYPE html>
<html lang="en">
<head>
    <meta charset="UTF-8">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <title>Hostname Input Form</title>
</head>
<body>
    <form action="/submit" method="post">
        <label for="hostname">Enter Hostname:</label>
        <input 
            type="text" 
            id="hostname" 
            name="hostname" 
            pattern="^[a-zA-Z0-9]+(-[a-zA-Z0-9]+)*$" 
            title="Hostnames can only contain alphanumeric characters and hyphens, but must not start or end with a hyphen." 
            placeholder="adsb-${MAC:02}" 
            value="adsb-${MAC:02}" 
            required>
        <button type="submit">Submit</button>
    </form>
</body>
</html>
```

I _think_ that's the correct client side validation for a hostname sans the special token.

Note that the _default_ value is `adsb-${MAC:02}` which uses the special `${varName:fmtModifier}` syntax to mean "the last 2 octets of the mac address". The idea here is that the _default_ value for this form should preserve the current "deterministically chosen hostname" behavior.

I absolutely would not fault you if you didn't want to add support for parsing the `${...}` syntax and instead went with a "just display the device MAC addresses on the page where the user can pick which octet(s) they want to manually include in the hostname" approach.

CoolNamesAllTaken, 2024-12-05:
I think that the easiest way to approach this would be to add it as a field to the AT+WIFI_STA command. My vague idea has been to eventually build a GUI that just sends AT commands to the TUI via the console WebSocket, but I'm still waiting for someone to tell me that it's a silly plan.

kquinsland, 2024-12-06:
> but I'm still waiting for someone to tell me that it's a silly plan.

If AT commands are _the_ API to `nvram` then it's a perfectly fine plan.

CoolNamesAllTaken, 2024-12-06:
Yep, they're the API of record. It keeps things simple and means that
anything can be configured automatically just as easily from serial, WiFi,
or ethernet. I haven't found anything wrong with it yet!

CoolNamesAllTaken, 2024-12-06:
@kquinsland Can you think of a reason that users might want a separate hostname for WiFi and for Ethernet? Or would you prefer that they be the same for simplicity?

CoolNamesAllTaken, 2024-12-06:
I'm going to use a single hostname for now to err on the side of less confusion. It will be applied to all interfaces and can be set with `AT+HOSTNAME=<hostname>`.

kquinsland, 2024-12-06:
> Can you think of a reason that users might want a separate hostname for WiFi and for Ethernet

I can't think of any good reasons that would also be "common". Nearly 100% of the time that you have some multi-honed device with different/redundant transports, you still only have **a** hostname and do aliasing @ the DNS layer.

FWIW, I honestly didn't expect support for WiFI and Eth @ the same time. This can complicate things if the "final" destination for traffic - the web server in this case, i guess - isn't aware of _which_ interface the request came in on.

CoolNamesAllTaken, 2024-12-12:
[image]
[image]

Resolved by https://github.com/CoolNamesAllTaken/adsbee/pull/48

closed 2024-12-12 by CoolNamesAllTaken