PantsForBirds/adsbee #45: FR: GPS/SD card support

kquinsland, 2024-12-05:
I know that basic GPS/NEMA support is already in flight (pun intended) but I was curious about using some of the spare GPIO to log data to a microSD card.

The primary usecase that I had in mind was for recording data in remote areas or with no real connectivity to anything else.
Think backpacking / offroading trips in the middle of nowhere.

Once back to civilization, you'd be able to take a look at the data on the SD card and see what rail numbers were overhead "after the fact".

[This post](https://old.reddit.com/r/esp32/comments/d71es9/a_breakdown_of_my_experience_trying_to_talk_to_an/) is old but through and has some basic performance numbers for the 1 bit and 4 bit modes versus the "regular" way of interfacing to an SD card. I'm not sure if it'd be possible to re-use the SPI bus between the 2040 and the ESP for "full speed" SD access and then you'd just need a single GPIO for the CS on the SD card or not.

CoolNamesAllTaken, 2024-12-05:
There's a  spare SPI port hanging off of the ESP32 that's currently intended for Ethernet support, I could imaging it being reused for SD card logging!

There's probably another discussion to be had about what kind of logging protocol would make the most sense. Storing raw packets at 100 packets / sec would take around 0.2GB / day so not the biggest deal, but worth thinking about.

kquinsland, 2024-12-06:
> There's a spare SPI port hanging off of the ESP32 that's currently intended for Ethernet support, I could imaging it being reused for SD card logging!

Oh right. I keep forgetting about that! I think using that in an either/or configuration is perfectly acceptable.  If you have eithernet, you also have a sd card... on the other side of the ethernet link. If you don't have ethernet, you can have a sd card locally.

> 100 packets / sec would take around 0.2GB / day so not the biggest deal, but worth thinking about.

Apparently there's a [library](https://github.com/greiman/SdFat) that adds `exFAT` support so multi-terabyte (yes, [really](https://www.amazon.com/TEAMGROUP-T-Create-Creators-Compatible-TTCS2TIA2V3003/dp/B0CYSTRR5T/)!) SD support should be possible.