PantsForBirds/adsbee #50: FR: Settings/Config dump/export

Wferr, 2024-12-15:
Would be nice to have a better/easier way to dump and re-apply settings between firmware updates.

AT+SETTINGS? Gives you all the settings but then you still need to re-craft the AT commands

What if there was a way for it to dump the settings in the AT command format so it could be copied and paste back into the updated board?

kquinsland, 2024-12-16:
Quite a few ways to skin this cat. The least painful way is to keep the firmware as simple as possible :).

Forward/backwards compat is ... tricky, at best. The Tasmota project does this really well but even they have limits (usually there's a migration path where you can't skip forward too many versions).
To get an idea of how complex their internal state store is, take a look at (one of) the tools that can decode it: https://github.com/tasmota/decode-config/blob/development/decode-config.py

You'd use this tool if you're ever _super screwed_ and need to parse out the original configuration from some backup (ask me how I know....).
Take a look at some of the bit twiddling and the like used to move settings around from version to version :exploding_head: .

Granted, the data/structure here is a _lot_ simpler, but it's still going to be a mess if you have to do it all in firmware.
I'd argue that a super simple API and a general "do the complicated stuff off device w/ something more accessible like python" policy would work well.

A simple/sane approach would be to implement a basic `AT+NVRAM_INFO` or similar that does one thing: it spits out the version of the storage layout (if different from fw) and all of the keys that it knows about.

Then it's up to the user to write a tool to save/convert/restore this.

e.g.:

```python
with open(/dev/ttyUSB0) as con:
  con.write(AT+NVRAM_INFO)
  device_info = con.read()  # Might return a string like `V1.2;278B;K: HOSTNAME, LOG_LEVEL, PROTOCOL ...`
                            # Version 1.0, consumes 278 bytes, keys that can be set are ....

  for k in device_info
       setting_value = con.write(f"AT+{k}?").read()
       print(f"The setting known as {k} has a value of {setting_value}")

```

Would produce output of:

`The setting known as LOG_LEVEL has a value of SILENT` 

which can then easily be put into a json/yaml/toml ... etc file for backup/modification/reload.

CoolNamesAllTaken, 2025-01-18:
It's finally time for me to do some work on this! I'll be targeting this feature for release as part of 0.7.1. I want to have this implemented before we have any new settings versions that would require users to re-enter their settings without assistance.

Taking into account the feedback I've received so far, my current plan is to have something like the `AT+SETTINGS?` command that puts out machine-readable text (i.e. AT command strings). So running `AT+SETTINGS?` would output the human-readable settings printout that we currently have, while running `AT+SETTINGS?AT` would output the settings as a series of AT command strings that the user could then paste into a text file. After upgrading the firmware, the user could input these AT command strings right back into the serial terminal to bring up the device.

Obviously this only works if the user can still get in touch with the device after the firmware upgrade, so there will need to be some concept of a "safe zone" in settings NVRAM that will never ever change format (in a non backwards compatible way) or be overwritten during a firmware update. This area would contain things like WiFi network name and credentials, network interface settings, and device hostname. I'm not sure whether I want to introduce this as part of 0.7.1 or as part of a future update.

If the settings "safe zone" was introduced later, users would need to connect to their device over the default connection methods (USB console or the ADSBee's self-published WiFi AP) to reload the settings, which isn't the end of the world but could be slightly annoying. What do y'all think?

CoolNamesAllTaken, 2025-01-18:
Switching to `AT+SETTINGS?DUMP` instead of `AT+SETTINGS?AT` since the extra AT prefix was triggering the AT command library (I should have forseen this, given that I wrote said library).

Seems to work OK!
[image]

CoolNamesAllTaken, 2025-01-19:
I'm adding an interface to the webpage to allow download and upload of settings as plaintext files.

[image]

[image]

closed 2025-01-19 by CoolNamesAllTaken