PantsForBirds/adsbee #59: Add aircraft ground position decoding on 1090MHz

CoolNamesAllTaken, 2025-01-19:
Currently, ADSBee only decodes airborne positions for aircraft. It would be good to also report aircraft on the ground along with their associated positions.

This will require a [slightly different CPR decoding implementation](https://mode-s.org/1090mhz/content/ads-b/4-surface-position.html) with a different global grid, as well as a reference position. The reference position could be pulled from some sort of averaged position across multiple airborne aircraft contacts, or from a position that can be input by the user.

VirusPilot, 2025-08-17:
> This will require a [slightly different CPR decoding implementation](https://mode-s.org/1090mhz/content/ads-b/4-surface-position.html) with a different global grid, as well as a reference position. [...]

If I read this correctly, the reference position can be the position of the receiver which is easy in case a GPS is attached. `readsb` has obviously implemented the CPR surface decoding here: [decodeCPRsurface](https://github.com/wiedehopf/readsb/blob/539e11d463153b08378c84cd5f5c8394592c9871/cpr.c#L223)

VirusPilot, 2025-08-19:
That is a great idea indeed!