PantsForBirds/adsbee #73: [FEATURE] TCP/UDP Serial Bridge to access ADSB Data in all available protocols over WiFi network

b14ckyy, 2025-02-12:
Since we have an ESP32 that can further buffer and distribute the ADSB Data to Feed online services already, it would be very nice to have the ability to access these data over wifi network from local devices. 

MWPTools for INAV supports udp and tcp connections to gather telemetry from aircraft over wifi and ADSB Mavlink1/2 is handled like any other Serial device feeding telemetry into it. So this would be great to have a single WiFi Network handling all the data streams. 
Also Mission Planner for Ardupilot supports ADSB over TCP as far as I understand it, some info I found here: https://discuss.ardupilot.org/t/ads-b-mission-planner-and-flightaware-ads-b-dongle/21362/9

CoolNamesAllTaken, 2025-02-14:
This is a good idea, and I think it could be implemented relatively easily by allowing MAVLINK to be selected as a feed protocol option on the ESP32. There would need to be a decent amount of refactoring in the backend to make it pretty, but I don't see anything that would prevent this from being implemented.

b14ckyy, 2025-02-14:
Yeah I think Mavlink makes most sense since ADSB-Mavlink 1 and 2 messages can be read by Mission Planner and MWPTools so it would work with both Ardupilot and INAV. Not sure about QGC.

pantsforbirds-claud, 2026-09-27:
THis has been implemented for a while, forgot to close the issue!

closed 2026-09-27 by pantsforbirds-claud