PantsForBirds/adsbee #78: Postion is sometimes wrong

error414, 2025-02-22:
Hi, 

I noticed that position of aircraft is sometimes wrong, it happens only for aircraft with weak signal. In the position history is visible that position was reported twice wrongly. 
55.00342","21.26767 and 55.00342","21.26767

it's not happen often, so I don't know if it's issue, or it's normal

[image]

[["48.93502","18.57673"],
["48.93683","18.57578"],
_**["55.00342","21.26767"],
["55.00658","21.26582"],**_
["49.06783","18.50770"],
["49.07025","18.50647"],
["49.07804","18.50246"],
["49.11052","18.48565"],
["49.11387","18.48392"],
["49.13356","18.47366"],
["49.14606","18.46710"],
["49.15274","18.46365"],
["49.15544","18.46216"],
["49.15645","18.46168"],
["49.16261","18.45848"],
["49.16519","18.45710"],
["49.16764","18.45580"],
["49.16853","18.45534"],
["49.17279","18.45309"],
["49.17448","18.45219"],
["49.17709","18.45082"],
["49.17806","18.45031"],
["49.18062","18.44900"],
["49.18680","18.44576"],
["49.19897","18.43940"],
["49.20511","18.43612"],
["49.20851","18.43434"],
["49.21193","18.43252"],
["49.21440","18.43126"],
["49.21628","18.43027"],
["49.21796","18.42935"],
["49.22186","18.42731"],
["49.22345","18.42646"],
["49.22690","18.42465"],
["49.23582","18.41999"],
["49.23679","18.41942"],
["49.23962","18.41794"],
["49.24063","18.41742"],
["49.24887","18.41309"],
["49.25920","18.40767"],
["49.26013","18.40724"],
["49.27233","18.40080"],
["49.27401","18.39993"],
["49.28087","18.39632"],
["49.28434","18.39451"],
["49.29496","18.38894"]]

CoolNamesAllTaken, 2025-02-22:
Thanks for flagging this! The new verified NASA CPR decoder is much better than my custom implementation, but I still haven't added a "reasonableness" check for the decoded coordinates, which should be the final piece of the puzzle that's required to prevent invalid decodes.

Are all of these incorrect positions way off, or are some of them more subtle? If they're way off, I could calculate the centroid of trusted positions from other aircraft and then try filtering based on that info.

CoolNamesAllTaken, 2025-02-23:
Great, that sounds like it should be quite filter-able then. I'll target this for the next release.

CoolNamesAllTaken, 2025-03-01:
Moving this issue to 0.7.5 since I'd like to cut 0.7.4 loose so that people can use the MAVLINK fixes.

I've done some thinking on filtering, and I think that the easiest way to implement a filter might be something like:
* Wait until the ADSBee sees N aircraft with valid positions.
* Convert aircraft coordinates from degrees to cartesian XYZ coordinates on a unit sphere, and calculate the centroid of those positions as an estimate of the receiver's position.
* Reject decoded positions that are more than X km away from the receiver's calculated position.

There is probably some debate to be had about whether aircraft positions that don't satisfy this filter should be reported, but with an invalid position flag, or whether they should be blocked out entirely.

If an aircraft transponder is reporting an incorrect position, and there isn't a CPR decode issue, it might be misleading for the ADSBee to refuse to report any position for the aircraft at all. I could see this being important in the context of debugging faulty equipment, for example. However, there are some protocols like GDL90 that don't support marking the aircraft as invalid, and many tools don't actually filter aircraft position based on the validity flags. For best default operation for most users, it probably makes sense to just hold the aircraft's last known position when receiving a packet with a garbage location.

error414, 2025-03-04:
that kind of filtering is already implemented in inav code for ADSB, Inav know his own position and  ignores aicraft too futher. 

Do you have idea why wrong possion passes through CRP decoding? Is there any kind of false positive CRC checking? 

I have new device so I did next tests

What is interesting, the wrong position is always somewhere near Itally. It seems like number before the decimal point is wrong. 
On the first image is 

OK position: 48.86365  / 16.72011
Wrong position: 42.80823 / 14.77586

And if you change 42 to 48 and 14 to 16 then GPS coordinates looks ok. Even if you look to other values before, decimal numbers have right descending / asceding trend. i don't think it's coicidence. 

exactly same situation on second screenshot. 

[image]  

[image]

[image]

CoolNamesAllTaken, 2025-03-13:
Sorry for the late reply! My original thought was that this could be due to aircraft hopping between latitude zones, and invalid decodes might be happening with one set of coordinates from each zone, but it appears that my code is already checking for the validity flag returned by the NASA CPR library.

[Valid flag set by checking latitude zone mismatch](https://github.com/CoolNamesAllTaken/adsbee/blob/main/firmware/common/adsb/nasa_cpr/airborne/cpr.c#L51)
[Decoder returns valid flag](https://github.com/CoolNamesAllTaken/adsbee/blob/main/firmware/common/adsb/nasa_cpr/nasa_cpr_airborne.cpp#L32)
[Dictionary filters update based on valid flag](https://github.com/CoolNamesAllTaken/adsbee/blob/main/firmware/common/adsb/aircraft_dictionary.cpp#L135)

[Mode S Riddle CPR Reference](https://mode-s.org/1090mhz/content/ads-b/3-airborne-position.html)

In this case it may be due to some other less common issue with CPR decoding--not sure what the exact mechanism is, but they do recommend filtering positions coming out of the CPR decoder to ensure validity.

error414, 2025-03-13:
Is it possible that there is issue with clearing flags? So far I don't manage compiling so I just have to use my imagination. 

If I go backwards, [aircraft_dictionary.cpp](https://github.com/CoolNamesAllTaken/adsbee/blob/main/firmware/common/adsb/aircraft_dictionary.cpp#L135) you set flag only if it passes through validation, so flash is not never set to 0. 

DecodePosition is called from https://github.com/CoolNamesAllTaken/adsbee/blob/main/firmware/common/adsb/aircraft_dictionary.cpp#L771 on the object aicfraft. 

Instance of object aicraft is taken from https://github.com/CoolNamesAllTaken/adsbee/blob/main/firmware/common/adsb/aircraft_dictionary.cpp#L356  by function GetAircraftPtr, the function get instance of object aicraft by icao or create new instance. 

And maybe it's the issue, once flag is set than never is reset.

CoolNamesAllTaken, 2025-03-13:
In this case I think the result_valid boolean is [set by the NASA CPR decode function](https://github.com/CoolNamesAllTaken/adsbee/blob/main/firmware/common/adsb/aircraft_dictionary.cpp#L124), so it should get freshly assigned to false or true during every decode.

The intended functionality is that if we decode an invalid location due to the aircraft crossing latitude zones, the track is not updated, and the last valid location is held.

The validity flags that are passed through via MAVLINK are in a different section of code, and they are one-way latching flags. That's to say, once a valid altitude / heading / position / etc is decoded, that validity flag is set to true and remains true until the aircraft is removed from the dictionary.

error414, 2025-03-15:
You are right. 

I managed to compile FW. Here is mention https://aviation.stackexchange.com/questions/87476/why-would-my-ads-b-airborne-position-decoding-sometimes-result-in-a-latitude-off that add/even massages should be delayed to each other max 10s. You use more complicated mechanism, there is no issue because I got wrong position and messages time beetween odd/even messages was 2s. 

So it seems that some kind of filters has to be used as you said.

I tried to prototype few types of filter. The best result I got with quite complcated filter, I used JS because it easy language for prototyping. 

https://github.com/error414/adsb-viewer/blob/main/src/service/csbeeParser.js#L70

1) calculate distance between two position
2) from the speed and time between messages calculate possible distance which aircraft would fly, and multiply it by magic constant to make some room.
3) compare distance from point 1 with point 2
4) if distance from point 1 is bigger then new position is ignored (for now) and it's needed confirmation by receiving next frame. 

[image]

CoolNamesAllTaken, 2025-03-16:
Awesome! I'm glad that you were able to get the firmware compiling :)

Your distance filter implementation is very clean and understandable--I
like it better than my original plan for a filter, which was based on
distance to a calculated receiver location.

In your testing, have the invalid locations only existed for a single
sample? If so, the double-check distance filter you have implemented would
probably be sufficient.

CoolNamesAllTaken, 2025-03-16:
I have a new branch for this filter development, I'll start cooking on some unit tests and an implementation of your velocity-based filter.

error414, 2025-03-16:
Cool, I did some more test and it seems that CPR decoding is correct, I used http://jasonplayne.com:8080/ to verify positions, or jasonplayne contains same bug :D

Maybe it could help, I captured messages with wrong positions. 

```
http://jasonplayne.com:8080/  I have no idea why but is seems that order for jasonplayne is "the most recent packet firs"

Help
odd \ even \ is odd last -> "8D48922358C387D91DF354DCCCB8" "8D48922358C38066020D8401D571" "0"
jasonplayne query: *8D48922358C38066020D8401D571; *8D48922358C387D91DF354DCCCB8;

1)
adsbbee: 48.5977 18.70521 
"8D48922358C387D91DF354DCCCB8" "8D48922358C38066020D8401D571" "0" (jasonplayne)-> "Lat": 48.59770202636719, "Lon": 18.705209585336537,
OK

2)
adsbbee: 54.68351 21.54819 !!!!
"8D48922358C387D91DF354DCCCB8" "8D48922358C3806C3E0C8BC657BB"  "1" (jasonplayne)-> "Lat": 54.683517197431144,  "Lon": 21.54818448153409,
NOK

3)
adsbbee: 48.64016 18.68485
"8D48922358C387E2E7F1D0A99112" "8D48922358C3806CEA0C714976EF" "1" (jasonplayne)-> "Lat": 48.640152236162606,  "Lon": 18.684853001644736,
OK

-------------------------------------

1) 
adsbbee: 49.30659 17.4134
"8D48C22D60AB0452BFAD19A695E0" "8D48C22D60AB00DEABC5DB78FCD6" "1" (jasonplayne)-> "Lat": 49.30659407276218, "Lon": 17.413402356599505,
OK

2) 
adsbbee: 43.35361 32.54067 !!!!
"8D48C22D60B104710F94F963E8B6" "8D48C22D60AB00E705C60B37E092" "0" (jasonplayne)-> "Lat": 43.353607177734375, "Lon": 32.540673544240555,
NOK

-------------------------------------
1) 
adsbbee: 48.7739 16.62398
"8D4D22B75859B7F959826F89952E" "8D4D22B75859C0845B9A0B7E00CD" "1" (jasonplayne)-> "Lat": 48.77389681541313, "Lon": 16.62397685803865,
OK

2)
adsbbee: 48.76593 16.62687
"8D4D22B7585987F8038297D4FF31" "8D4D22B758599083419A2DF64613" "1"  (jasonplayne)->  "Lat": 48.76593638274629, "Lon": 16.626867997018913,
OK

3) 
adsbbee: 54.80888 19.16881 !!!!
"8D4D22B7585727EE2783A8627A77" "8D4D22B758597081BD9A58249C01" "1"  (jasonplayne)->  "Lat": 54.80888237387447, "Lon": 19.168812144886363,
NOK

4) 
adsbbee: 48.69054 16.65217
"8D4D22B75855D7EBDF83E7705A0A" "8F4D22B75855C075DB9BA444F331" "0" (jasonplayne)-> "Lat": 48.69053649902344, "Lon": 16.652174729567307,
OK
```

CoolNamesAllTaken, 2025-03-22:
Starting work on this now--I think that in order to make calculations fast for this position screening, it will be best to utilize a fixed-point format like the Alternative Weighted Binary used in the CPR decoder library, and use a number of lookup tables to enable fast calculation of haversine and asin values. Currently working on a script to generate the tables that will be required for this.

CoolNamesAllTaken, 2025-03-25:
The position filter is implemented in [this PR](https://github.com/CoolNamesAllTaken/adsbee/pull/83), and is currently available for testing in the 0.7.5-rc3 release! I'd love some help with testing it out if you have a moment 😄 

It ends up using a combination of custom generated lookup tables (some of which are quite large) as well as fixed and floating point math. Enabling the filter adds about 20-30us to the packet ingestion time when running on the device. Hopefully this doesn't cause much of an issue since the total time to ingest a position packet is under 100us.

I welcome any ideas for optimizing the code further (there are still a few very expensive floating point operations being done, since I didn't want to deal with 64 bit fixed point math). We can also look at disabling the CPR filter under certain conditions to try to avoid causing problems with feeder applications that don't need the extra filtering. 

[image]

error414, 2025-03-25:
I tested new FW 0.7.5 RC 3. 

1) filter works, 2 hours of running and none of wrong position, in prev version I got wrong position in 20 min +-
2) sensitivity is slightly worse than version 0.7.4, but surprisingly sensitivity of 0.7.5 is still better than aeromax tt-sc1

So I would say, it's success :). I see in code you prepared functionality for filter disabling, I can imagine to add it to AT commands. 

I'm impressed you were able to make filtering as fast.

CoolNamesAllTaken, 2025-03-26:
It's great to hear that the filter seems to be working!

Do you have any additional info about the reduced sensitivity? I'm surprised to hear about that, given that I didn't change any of the demodulator code.

It's possible that core 0 is getting too heavily loaded, and its loop time is getting long enough that some packets are dropped from the queue. Do you see any errors on the serial terminal?

error414, 2025-03-26:
About sensitivity, I did not measure it precise, it's just feeling, I saw many planes where was not filled callsign of plane. For precise measuring I would have to compare it with 0.7.4 side by side. Would you like to test it? I would have to uninstall adsb from plane :), it's a reasson why I didn't test do it. 

I thought it's caused by filtering. Sensitivity was still better than aeromax so I did not give much attention.

CoolNamesAllTaken, 2025-04-12:
Fixed by https://github.com/CoolNamesAllTaken/adsbee/pull/83

closed 2025-04-12 by CoolNamesAllTaken